procount/pinn #852: Feature request: Alpine Linux

silviu-social1st-ro, 2024-11-26:
Hello, i am running an RPI 3b+ and Alpine Linux save it from the trash bin...

It runs super fast, makes it feel snappy again.
I have dockers on it for homeassistant, zigbee2mqtt, mqtt, mariadb and a wordpress/redis stack :)

just discovered your project, do yout think it would be possible to integrate alpine? Or are there instructions so I could give a hand with Alpine? Thanks!

procount, 2024-11-26:
I believe Alpine is already included, although I'm not sure for which RPi models off the top of mry head.

procount, 2024-11-26:
Alpine v3.8.2 is included for the 3B+. Its a bit old from 2018 so probably needs updating. Probably you can do that manually after installation.

silviu-social1st-ro, 2024-11-26:
Thanks! i was documenting before installing PINN and I couldn't see Alpine in any YT tutorials. I will test tomorrow.

closed 2024-11-26 by silviu-social1st-ro