procount/pinn #859: Fresh Install Not Booting

Jarod96, 2024-12-23:
Hi, I've install PINN p3.9.3 on a 64gb SD Card using RPi imager but RPi 4B 4G don't show anything past the initial rainbow splashscreen.
So I downgraded to p3.8.8 and it work again, I have the PINN menu and a popup for an update.
(ethernet cable + wireless keyboard)
Once the update finish, the RPi don't show anything again after the splashscreen, the green light flash frequently at first then just some deemed blinking occurs at a regular interval.

Jarod96, 2024-12-23:
Weird how a not-fully inserted HDMI cable can make a difference between two different version of the same OS.

closed 2024-12-23 by Jarod96

procount, 2024-12-23:
lol 😄 
Yes, Just like the latest Raspberry Pi OS, the new KMS video drivers are a bit picky about reading the EDID information from the monitor.