procount/pinn #861: Data partition size

JanSykor, 2024-12-25:
Hello @procount 
Is there aby reason to keep Data partition in Pinn on 500MB? On new installation it was possible to change of size of all systems but not Data partition size.
I bought 512 GB raspberry SSD with M2 hat and want keep rest of disk for data, but because it was not possible to change Data size, I am shortening last project space to 10GB and from rest made new ext4 partition for data. After this I can not boot my both raspios systems. IT stacks on some message about missing partition please sebe attached picture. Do you have some idea how to fix this problem?
Strange is that LibreElec working without problem.
[image]
Thanks for your help.
Jan

procount, 2024-12-26:
By altering the partitions manually, you have probably inadvertently changed the diskID. Some tools like Parted may do this. As you are using an SSD, it will be using partuiid references based on the diskID, so it won't find the partitions. (LibreELEC may use labels or another method to reference its partitions - just a guess - I haven't checked)
Posting the output of `fdisk -l` and the contents of /settings/installed_os.json  or the cmdline.txt of an OS that will not boot will confirm this. (/settings refers to partition No. 5 when mounted in PINN).
Restoring the diskID to that referenced in installed_os.json will probably fix it. (Just ask if you don't know how to do this).

Regarding the data partition size, this can now be adjusted on installation when the Adjust Size of OSes dialog comes up. Even though the data partition is indicated as having a fixed size, this just means any spare capacity will not be shared to it, but you can add any extra size you want. But first you must remove the extra allocations from the other OSes to make room.
I suggest clicking the Clear button, entering the extra MBs you want to add to the Data Partition, then click the Balance button to share out any remaining space among the other expanding partitions. Of course you can manually set the extra sizes for the other OSes instead if you wish.

JanSykor, 2024-12-27:
Hello @procount 
I am sending requested file in zip - I am not sure what is wrong.
If we did not fix it, I will try your sugestion with Clear and reinstall
Thanks for your help
Jan
[pack.zip](https://github.com/user-attachments/files/18262740/pack.zip)

procount, 2024-12-28:
The fdisk command in Raspios initramfs is not very comprehensive and does not show the diskID.
Could you repeat it from PINN's command shell?
Boot into PINN (press shift if necessary to remain in PINN)
Ctrl-Alt-F2 to get to the command prompt.
Login with username/password of root/raspberry
Capture the output of `fdisk -l`

Alternatively, if you have enabled PINN's ssh, you can login remotely with a PuTTY and capture the output that way.

procount, 2024-12-28:
Yeah, you see the current disk Identifier from fdisk is 0x6d144741, but in the cmdline.txt files, it is expecting it to be 7cde0769. So changing the partition sizes has changed the diskID.

You can change the diskID back again using fdisk.
```
fdisk /dev/nvme0n1
x
i
7cde0769
r
w
q
```

JanSykor, 2024-12-28:
Yes you have right. Now both system is booting and in pinn fdisk is correct diskID.
Just the name must be in format 0x7cde0769 and command w not working, but after restart it was ok.
[image]

Unfortunately lite system cannot see other drives, but I think this is not related to pinn

Anyway thank you very much for your quick help
Jan

buvinghausen, 2025-01-28:
@procount I'm assuming the preferred workaround is to just allocate less space to each OS leaving free space at the end of the drive which you can then go in and manually create the data partition which presumably wouldn't mess with any of the drive ids?

buvinghausen, 2025-01-28:
@procount the UI states the data partition is a fixed size disk and won't let you expand it past the 512 gb amount.  I wound up just leaving a bunch of free space and it appears pinn makes a share partition with the balance (wihch was ultimately what I was after)

procount, 2025-01-28:
Please see the post above: https://github.com/procount/pinn/issues/861#issuecomment-2562044478

> Regarding the data partition size, this can now be adjusted on installation when the Adjust Size of OSes dialog comes up. Even though the data partition is indicated as having a fixed size, this [...]
I suggest clicking the Clear button, entering the extra MBs you want to add to the Data Partition, then click the Balance button to share out any remaining space among the other expanding partitions. Of course you can manually set the extra sizes for the other OSes instead if you wish.

But if you've managed it another way, then that's fine. Note that PINN won't create additional partitions. It just shares the spare capacity out amongst the dynamic partitions of all the OSes.