procount/pinn #862: Page name translated partialy

JanSykor, 2024-12-28:
**Describe the bug**
After system reboot is page name only partially translated - please see red frame in the pictures
After change to english and back to Czech the translation is correct, but after next reboot is partialy translated again
I am looking to the code, but cannot find problem
[image]
[image]
[image]

**To reproduce**
After change to english and back to Czech the translation is correct, and reboot - is partialy translated.

**Expected behaviour**
Keep translation same after reboot and after language change

**Actual behaviour**

**System**
Add answers to the following questions:
Pi 5 with raspberry SSD - 512 GB
Pin 3.9.3
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which version of PINN?

**Logs**
If applicable, add the relevant output from `/tmp/debug` or `dmesg`.

**Additional context**
Add any other relevant context for the problem.

Thanks for your help
Jan

procount, 2024-12-29:
Thanks for reporting. This will require some debugging.

procount, 2025-01-06:
Problem found. Will be fixed in next release.

commit 2d4158ee (2025-01-07): Fixes #862 - partial tab translations
closed 2025-01-07 by procount