procount/pinn #866: Freenove FNK0078 not supported anymore

MishterKirby, 2025-01-02:
**Describe the bug**
Screen shows test pattern/otherwise stays uninitialized upon bootup after rainbow screen.

**To reproduce**
Boot Raspberry Pi with screen connected to DSI connector.

**Expected behaviour**
PINN should show up on screen after a few seconds.

**Actual behaviour**
After rainbow screen, screen goes blank for a few seconds before cycling between colors. Pi is otherwise unresponsive.

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? 
Raspberry Pi 4B
* Which version of PINN?
PINN 3.9.3

**Logs**
If applicable, add the relevant output from `/tmp/debug` or `dmesg`.
Not possible

**Additional context**
I haven't tried yet, but I'm assuming HDMI is still working fine and should display an image.

procount, 2025-01-02:
The latest firmware in 3.9.3 seems to select KMS for the video drivers.
PINN does not support DSI/DPI on KMS yet, but you should be able to get it working by manually adding the appropriate dtoverlay with config.txt settings, and maybe a video setting in cmdline.txt

MishterKirby, 2025-01-03:
I see, I'll try later and report back.

Would that also explain why some of the operating systems I tried installing (such as Konsta's AOSP/Lineage OS) also displayed the same behavior on 3.8.x?

procount, 2025-01-04:
Only Raspios OS will "just work" with their own displays. Other OSes may need separate configuration. See the official documentation for details.
For Konsta's Lineage - there are instructions on how to configure it for the 7" display on his FAQ.

MishterKirby, 2025-01-04:
Alright, I managed to get PINN working with the config.txt changes, so I'll close this issue.

closed 2025-01-04 by MishterKirby

strife-py, 2026-05-11:
> Alright, I managed to get PINN working with the config.txt changes, so I'll close this issue.

I'm having the same issue with that touchscreen. What did you change in the config.txt to make it work?

strife-py, 2026-05-13:
As requested...

**Full dmesg output:**

```
[ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[ 0.000000] Linux version 6.6.51-v8 (kevin@ubuntu) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot p3.9.2-4-g386d682d) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT Mon Nov 4 14:58:29 GMT 2024
[ 0.000000] KASLR enabled
[ 0.000000] random: crng init done
[ 0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[ 0.000000] efi: UEFI not found.
[ 0.000000] Reserved memory: created CMA memory pool at 0x000000000d000000, size 512 MiB
[ 0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[ 0.000000] OF: reserved mem: 0x000000000d000000..0x000000002cffffff (524288 KiB) map reusable linux,cma
[ 0.000000] OF: reserved mem: 0x000000003f1759e0..0x000000003f175a15 (0 KiB) nomap non-reusable nvram@0
[ 0.000000] Zone ranges:
[ 0.000000] DMA [mem 0x0000000000000000-0x000000003fffffff]
[ 0.000000] DMA32 [mem 0x0000000040000000-0x00000000fbffffff]
[ 0.000000] Normal empty
[ 0.000000] Movable zone start for each node
[ 0.000000] Early memory node ranges
[ 0.000000] node 0: [mem 0x0000000000000000-0x000000003e5fffff]
[ 0.000000] node 0: [mem 0x0000000040000000-0x00000000fbffffff]
[ 0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[ 0.000000] On node 0, zone DMA32: 6656 pages in unavailable ranges
[ 0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[ 0.000000] percpu: Embedded 30 pages/cpu s85352 r8192 d29336 u122880
[ 0.000000] pcpu-alloc: s85352 r8192 d29336 u122880 alloc=30*4096
[ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[ 0.000000] Detected PIPT I-cache on CPU0
[ 0.000000] CPU features: detected: Spectre-v2
[ 0.000000] CPU features: detected: Spectre-v3a
[ 0.000000] CPU features: detected: Spectre-v4
[ 0.000000] CPU features: detected: Spectre-BHB
[ 0.000000] CPU features: kernel page table isolation forced ON by KASLR
[ 0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[ 0.000000] CPU features: detected: ARM erratum 1742098
[ 0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[ 0.000000] alternatives: applying boot alternatives
[ 0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 clk_ignore_unused snd_bcm2835.enable_hdmi=0 smsc95xx.macaddr=E4:5F:01:AE:CD:A6 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000 quiet ramdisk_size=65536 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json loglevel=2 sdhci.debug_quirks2=4
[ 0.000000] Kernel parameter elevator= does not have any effect anymore.
Please use sysfs to set IO scheduler for individual devices.
[ 0.000000] Unknown kernel command line parameters "repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json", will be passed to user space.
[ 0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[ 0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 1009408
[ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[ 0.000000] software IO TLB: area num 4.
[ 0.000000] software IO TLB: mapped [mem 0x000000003a600000-0x000000003e600000] (64MB)
[ 0.000000] Memory: 3366600K/4102144K available (17472K kernel code, 2362K rwdata, 5936K rodata, 5632K init, 1124K bss, 211256K reserved, 524288K cma-reserved)
[ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[ 0.000000] ftrace: allocating 53466 entries in 209 pages
[ 0.000000] ftrace: allocated 209 pages with 4 groups
[ 0.000000] trace event string verifier disabled
[ 0.000000] rcu: Preemptible hierarchical RCU implementation.
[ 0.000000] rcu: RCU event tracing is enabled.
[ 0.000000] rcu: RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[ 0.000000] Trampoline variant of Tasks RCU enabled.
[ 0.000000] Rude variant of Tasks RCU enabled.
[ 0.000000] Tracing variant of Tasks RCU enabled.
[ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[ 0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[ 0.000000] Root IRQ handler: gic_handle_irq
[ 0.000000] GIC: Using split EOI/Deactivate mode
[ 0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[ 0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[ 0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[ 0.000364] Console: colour dummy device 80x25
[... 331 lines cut ...]
[ 3.498655] sd 0:0:0:0: [sda] 60125184 512-byte logical blocks: (30.8 GB/28.7 GiB)
[ 3.499978] sd 0:0:0:0: [sda] Write Protect is off
[ 3.500000] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[ 3.500808] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 3.508043] sda: sda1
[ 3.508430] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 4.701777] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
[ 4.702540] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan 4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[... cut ...]
[ 10.881817] EXT4-fs (mmcblk0p5): unmounting filesystem 208c979a-535a-4f50-87fd-8056cbe37bdd.
[ 11.748831] i2c 10-005d: deferred probe pending
[ 11.748870] platform 100000002.touch_reg: deferred probe pending
[ 14.947346] EXT4-fs (mmcblk0p5): mounted filesystem 208c979a-535a-4f50-87fd-8056cbe37bdd ro with ordered data mode. Quota mode: none.
[ 15.913694] logitech-hidpp-device 0003:046D:404D.0004: HID++ 4.1 device connected.
```

**/tmp/debug output:**
(empty)

Let me know if there's anything else I can provide. Thanks again!