procount/pinn #872: [feature request] berryboot format support or add openfde

thetechstoner, 2025-01-17:
Berryboot format support would be great if it could be added to PINN

Since it is so easy to create berryboot images

Been using fydeos and openfde 
Both have android and Linux app support
I know fydeos is not supported
But if can't add berryboot format support
Would like openfde added to PINN 
If you could

thetechstoner, 2025-01-17:
I think it would be cool to
train a Local ai model
to convert raspberry pi image to PINN format

thetechstoner, 2025-01-17:
AI could assist by generating conversion scripts or adapting metadata files to bridge these differences from berryboot and PINN format

lurch, 2025-01-18:
I did a quick search and https://github.com/maxnet/berryboot hasn't been modified for 4 years, so it probably doesn't even work on the Raspberry Pi 5?

actraiser30, 2025-01-18:
That's a shame berryboot is way better than pinn
Faster, lightest, high compatibility,small size images,huge image collection maintained by AlexG.

thetechstoner, 2025-01-19:
Interesting multi boot method
Using multiboot script

[MultiBoot for the Raspberry Pi 5, Raspberry Pi 4, and Raspberry Pi 3 - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?t=282912)

closed 2026-10-11 by thetechstoner