procount/pinn #873: Can't restore ROMs and other data to Batocera with PINN

appscaptain, 2025-01-20:
I had a working Batocera install. To be able to run both Kodi, Batocera and Raspian I wanted to reinstall using PINN (SD + USB HDD). The install completed with no error, however when I copied the ROMS, Saves and BIOS folders to the SHARE partition on the external USB harddrive used with PINN Batocera does not see the files even after scanning for new gamedata (Batocera settings). This works fine if I don't use PINN.

How do I add new data to Batocera with PINN so it recognises it?

Alternatively is there a way to move an existing entire Batocera partition into a PINN partition?

I am on a Raspberry Pi 3B.

Not sure what I am missing?

appscaptain, 2025-01-25:
Any loose suggestions/ideas?

procount, 2025-01-25:
I'm sorry, I have no idea about how Batocera references its data files. Using PINN should not make any difference, as it just facilitates moving Batocera to other partitions. Referencing external USB drives should be exactly the same.

appscaptain, 2025-01-25:
Ok, thanks! So where does PINN put the SHARE/userdata partition when you install on SD + HDD? I see the SHARE partition created on the HDD but moving files there does nothing (which it does if I do not use PINN but install BATOCERA normally).

See here under "Accessing Batocera's drive to add files"
https://wiki.batocera.org/add_games_bios

I am looking for this userdata partition under PINN.

appscaptain, 2025-01-25:
Ok, doing some more investigation it seems that:
When PINN steps up Batocera with a SD (16 GB) + HDD (500 GB) setup:
1. Batocera is not set correctly up to use the share partition on the HDD for userdata. Currently it chooses the SD card as the storage which must be a bug because Batocera was specifically set up to be installed on the HDD (the HDD > SHARE partition is created by Pinn but it is not selected inside Batocera). Storage disk can be set in System Settings inside Batocera but this doesn't solve anything because of 2 and 3 below.
2. The permissions for the SHARE partition contents seem to require root access so Batocera cannot read/write anything when the HDD > SHARE partition is selected inside Batocera settings. This means that the 500 GB HDD that was supposed to be used for storage for Batocera cannot be used at all. Only the the small partition with around 300 MB left on the SD can be used. Not sure how to change this. I tried reformatting the partition which doesn't help.
3. When accessing the SHARE partiton over the network it accesses the small one on the SD card with only about 300 MB free space, instead of the big 500 GB SHARE partition on the external HDD. Not sure how to change this.

So while 1 can be easily done, 2 and 3 I can't seem to solve. It seems like PINN doesn't fully set up Batocera settings correctly when it was supposed to put the storage partitions on the HDD.

An out of the box Batocera image doesn't have these issues. 

Any ideas on how to fix it?

procount, 2025-01-25:
Did you have PINN installed on the SD card?
Did you have the 500GB HDD plugged in when you installed Batocera?
When you installed Batocera with PINN, did you select the SD card or the HDD as the target drive?

procount, 2025-01-25:
PINN installs a standard image of Batocera  which consists of an unusually large boot partition (because it contains all of Batocera) and a storage partition for the games. PINN knows nothing about how Batocera can be configured for external drives as that is the concern of Batocera.

DO NOT  choose the option in PINN to install Batocera to HDD as this is probably not what you want. I understand you want to install 2 or 3 OSes, so by choosing the HDD, you will install all of their boot partitions on the SD card and their other partitions (rootfs or storage) on the HDD. This option was from before one could boot an RPi from USB drives and could only boot it from the SD card.

I found this [article ](https://wiki.batocera.org/store_games_on_a_second_usb_sata_drive) which explains how to change Batocera from using the internal storage to external USB storage. Set this up however you need to (permissions etc) for Batocera.

appscaptain, 2025-01-26:
Hi,

To answer you questions:
A. Yes PINN was installed on a FAT32 partition on the SD Card.
B. Yes, I had the 500 GB HDD attached during PINN installation and I chose it in the selection in the bottom of the install screen.
C. I selected the HDD as mentioned in B. above -  but are you referring to some other selection here? I only saw one time during the install process I could select the external HDD and that's what I did in step B.

For your second message, correct I want to have a much as possible on the HDD instead of the SD card (at least storage for Kodi and Batocera). So yes I choose the HDD in every part of the PINN install.

I am on a Raspberry Pi 3 B, so are you saying that the way to do this would be to:
1. Install Raspian to a SD first and set the program_usb_boot_mode=1 in  /boot/config.txt to set the OTP bit.
2. Copy the PINN installer to a HDD and boot it from that HDD (which I guess should be possible without even an SD card installed)?

About the article you link to this is exactly what I tried in step 1 I above but Batocera can't read the data on the HDD due to a permissions issue which I couldn't figure out how to fix no matter what I tried.

procount, 2025-01-26:
I think installing PINN to the HDD as you describe would be a better solution. You can use the RPi Imaging Utility in Raspberry Pi OS to install PINN directly from the internet to your HDD (you'll find PINN under the Misc Utility category). 
Remove the SD card and PINN should boot from the HDD.
In the latest versions of PINN, after selecting the OSes to install, you can adjust the amount of disk space used by each OS. You might want to give more space to Batocera for your games, or Kodi for any movies etc.

appscaptain, 2025-01-31:
Simply installing PINN to the USB (neither by using RPI Imager or by copying the files from Github) does not work on the Raspberry 3B. It doesn't boot and no image at all.

Even after setting the OTP bit and verifying that it is set correctly as described here:
https://brezular.com/2024/04/05/booting-raspberry-pi-3-b-from-usb-drive/

cgencmd otp_dump | grep 17: returns 17:3020000a as it should when USB boot is enabled.

Removing the SD card just leaves the Pi at a red light with no activity on the external. And having the SD installed gives a few green blinks every once in a while and some blinks on the external HDD for a while but no image.

I have been able to boot Raspbian from USB before with no issues.

procount, 2025-01-31:
PINN boots fine for me from USB on a 3B. I'm doing some work on a 3B at the moment so I was able to recheck it.
You seem to have enabled USB boot ok, so I'm not sure what the problem is.
I test this by pulling the SD card from the slot, inserting it into a USB card reader and plugging that into a USB port and booting it.
I have found some USB card readers can't be booted from, in which case, changing it for another usually works.
But in your case you're using a 500GB HDD, so that shouldn't be the problem.

Does your HDD get its power from the Pi, or do you use a powered hub? Depending on the HDD, the initial power required might exceed the PI's capability, in which case using a powered hub might solve the problem.

Alternatively, you could keep PINN on the SD card and install all your OSes to the SD card. Then follow the previous article about switching your Batocera storage to an external drive. If you still can't read anything from the HDD, then I would suggest a power issue again.

appscaptain, 2025-01-31:
Hi, 

The 2.5" USB HDD has a separate external official Raspberry Pi Power adapter. And the same power adapter worked fine with PINN when I used the disk with PINN with the previous SD+HDD setup. I don't hear the disk failing to spin up either or click either. So I think the power is ok.

Changing Batocera to use the external HDD storage using those exact instructions I tried earlier (see https://github.com/procount/pinn/issues/873#issuecomment-2613957856), but I had the problem with Batocera not being able to access anything on that drive (I think due to permissions). Not sure how to fix this and I guess it has something to do with how Batocera mounts the drive.

Not sure what more to try...

appscaptain, 2025-02-01:
Further update:
I tried 3 USB 2.5" enclosures (2.0 and 3.0 ) and 3 chargers. No change..

And then I tried to boot PINN from a 128 GB USB stick and install it there. It booted and installed on the USB, but only with an SD card installed with the bootcode.bin.

However after installation was succesful and I restart as prompted it shows the PINN prompt (Hold down shift to go to Recovery etc) and when I don't press anything it shows a black screen and the SD card blinks green 4 times, pauses a few sec. and repeats while the red LED stays on. There is no LED on the thumbdrive so don't know what that's doing.

If I go into PINN recovery and choose Boot there the same issue happens. Just a blank screen and the blinking LEDs in a similar way. It doesn't matter if I try to boot Batocera, Kodi or Raspbian. Same issue. 

If I completely remove the SD card as always the red just stays red and nothing happens.

Installing everything to the SD card works fully but too slow and with not enough space for it to be usable..