procount/pinn #875: Troubleshooting - typo in code block in instructions to capture debug logs?

mrengy, 2025-02-01:
---
name: Bug report
about: Create a report to help us fix your issue

---

**Describe the bug**
On https://github.com/procount/pinn/wiki/Troubleshooting , there's this code block

```
cd \tmp
mount -o remount,rw /mnt
cp debug /mnt/debug.txt
dmesg >/mnt/dmesg.txt
sync
```

When I type `cd \tmp` and then `enter`, I get `~sh: cd: can't cd to tmp`. When I type `cd /tmp`, it seems to work, and I was able to follow the rest of the lines to get the dmesg.txt file populated with logs.

Is `cd \tmp` a typo?

procount, 2025-02-02:
Yes. Thanks, I've corrected it.

procount, 2025-02-02:
This should create 2 files: dmesg.txt and debug.txt.
In #876 you only included dmesg.txt and omitted debug.txt

mrengy, 2025-02-09:
Thanks for pointing out the two files. I forgot about the other one.

closed 2025-02-09 by mrengy