procount/pinn #876: PINN continually reboots, won't load my installed operating systems (default one is RetroPie)

mrengy, 2025-02-01:
---
name: Bug report
about: Create a report to help us fix your issue

---

**Describe the bug**
PINN is continually rebooting. [This happens when I press nothing -screen recording](https://youtu.be/RNhUKm2RfOA). 

It also happens [when I manually select RetroPie from the menu and click "boot" - screen recording](https://youtu.be/g8HyW7fLgb8).

I also went to the [fix menu for RetroPie (the default boot OS), selected all 4 checkboxes, and everything seemed normal, I think - screen recording](https://youtu.be/cra-JpgLrX0).

**To reproduce**
Plug in the RPi and wait. 

**Expected behaviour**
After the timeout, RetroPie loads

**Actual behaviour**
PINN reboots infinitely, never loading an operating system.

**System**
Add answers to the following questions:

* Which model of Raspberry Pi?  Pi4B
* Which version of PINN? 3.9.2, I think - log file shows "(Buildroot p3.9.2-4-g386d682d)"

**Logs**
[Log file on pastebin](https://pastebin.com/ZLc2JnV6)

**Additional context**

cmdline.txt contains the following:

> forcetrigger remotetimeout=20 vncshare ssh quiet ramdisk_size=65536 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline 
repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json loglevel=2 
sdhci.debug_quirks2=4

procount, 2025-02-02:
There are a few things I can suggest:
1. PINN v3.9.2 has a small mistake in its configuration, which is normally benign, but please see https://forums.raspberrypi.com/viewtopic.php?p=2290901#p2290901 for a small recommended fix.

There may be a problem with your bootloader as I see it is dated from 2020. I have only seen such repeated bootloops on a Pi5 which was due to an omission in the early firmware (see https://forums.raspberrypi.com/viewtopic.php?p=2173713#p2173713 for details), but you can do some quick tests as follows:

2. When the PINN menu appears, move your mouse to cancel the remotetimeout timer you have set, then press Ctrl-Alt-F2 to get to the command shell. Login with root/raspberry then enter `rebootp 6` and see if your Pi boots into Retropie or PINN.
2. A further test is to create a file on PINN's recovery partition (/dev/mmcblk0p1) called `autoboot.txt`. Add one single line to that file that says: `boot_partition=6`. When you next boot your RPi, it should bypass PINN altogether and boot straight into Retropie. Does that happen? Please delete autoboot.txt afterwards.
3. Finally, once you have completed those tests, please see https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-boot-eeprom for details on how you can update your bootloader to the latest version. You'll need a spare SD card, and possibly an installation of raspios-lite, depending on your choice of method to use.

mrengy, 2025-02-09:
Thanks a lot for the support.

1. I followed that step in [tvservice is not supported…](https://forums.raspberrypi.com/viewtopic.php?p=2290901#p2290901) and edited config.txt accordingly. It cleared that error about tvservice I was seeing before.
2. I entered `rebootp 6` in the PINN command shell, and the Pi booted to PINN and continued to infinitely reboot to PINN like it had been doing. 
3. When I created `autoboot.txt` with that line `boot_partition=6`, it did bypass PINN and boot straight into RetroPie. I deleted `autoboot.txt.`
4. I'm backing up another SD card that I might want the data from and next will follow those instructions to update the bootloader. 

Do the results of the tests from steps 2 and 3 indicate I should do anything different?

mrengy, 2025-02-09:
4. I [updated the bootloader using a spare SD card.](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#imager) Got a green screen on the display, so I think it worked.

With the SD card that had PINN on it back in the RPi, I'm still getting the [infinite rebooting of PINN.](https://youtube.com/shorts/vOC1AHqSGr0?feature=share) Updated [debug.txt](https://pastebin.com/6FnxCxKc) and [dmesg.txt](https://pastebin.com/21YRi0Eb)

mrengy, 2025-02-09:
Should I just wipe my SD card and reinstall PINN and my operating systems within it? Or is something still messed up with my bootloader or something that would cause the problem to recur? 

At one point PINN was working and loading the operating systems. Not sure what happened to cause this infinite PINN rebooting bug.

procount, 2025-02-09:
> Not sure what happened to cause this infinite PINN rebooting bug. It started after I edited cmdline.txt  to try to get ssh and vnc enabled and force a delay so that I could choose a different OS from my phone over VNC on initial boot.

Have you by chance deleted the `sdhci.debug_quirks2=4` command from cmdline.txt?
Also, all parameters in cmdline.txt should just be on the one first line. Anything after a carriage return will be ignored.

mrengy, 2025-02-09:
I can double check later, but 90% sure my cmdline.txt file is exactly as is in the original post. And I don’t think there are any carriage returns. Is there anything worth keeping for the sake of trying to identify the cause of the bug more precisely? I think the easiest path would be to wipe and reinstall PINN at this point unless you have other ideas. -- 
Mike Eng (he/him)mike-eng.com
linkedin.com/in/mikeeng401

procount, 2025-02-09:
Try booting your Pi with no SD card. After a few seconds it should display the diagnostic screen.
Please post the first 3 lines at the top of the screen to the right of the Raspberry logo to confirm your bootloader has indeed been updated.

If you try reinstalling PINN, remember to change config.txt to avoid the tvservice error.

To edit config.txt or cmdline.txt. I suggest doing as follows:

1. Press the `m` key twice to get to the maintenance menu. The top entry for PINN should already be highlighted.
2. Press the `e` key to Edit the Config. There are 2 tabs here: one for config.txt and the other for cmdline.txt. You can edit config.txt here for the Tvservice issue. To alter cmdline.txt, I recommend pressing `Alt+e` to Edit the Options. This is a safer GUI way to edit cmdline.txt

mrengy, 2025-02-19:
Here are the first 3 lines I got when booting with no SD card. 

> Raspberry Pi 4 Model B - 4GB
> bootloader: 8ba17717 2023/01/11
> update-ts: 1674059736

[image]

procount, 2025-02-20:
Your bootloader still looks rather old at 2 years, but I haven't checked what the latest version is.
Not sure what to suggest at the moment, apart from reinstalling PINN.