procount/pinn #879: Batocera does not install  "OS(es) installed with errors. See debug log for details."

appscaptain, 2025-02-09:
Hi;

Installed the newest PINN on an SD card (no USB disk) using a PI3B. Whenever I include Batocera in the installation I get "OS(es) installed with errors. See debug log for details." It just quickly skips it whenever it gets to it. The internet is fine (the other distros install fine).

Using ctrl-alt-F2 does not bring me to the console so I am not sure how to read the log. And I don't see it on the SD card it when I plug it into a linux computer.

I am installing to a 32 GB SD card. And tried two other 16 GB SD cards as well. All do the same. DietPi, RaspbiOS, LibreElect etc. install fine. It's only Batocera that errors out. No matter if I give it 14 GB or only installs it.

Any ideas?

appscaptain, 2025-02-09:
Ok, I think I found the log now inside the GUI. It's weird I installed Batocera but I see it trying to get recalbox7-rpi3 and not Batocera?

Also see quite a few "libpng warning: Unknownc iTXt compression type or method" errors

And the length seems to be only 160 for recalbox7-rpi3 as well.

This should be easy to reproduce. Simply try to install batocera and it will fail like this instantly.

Any ideas how I can fix it?

procount, 2025-02-09:
It seems the maintainers of Batocera have changed their download locations on their server, so the file cannot currently be found.

I will have to update all the details to make it available again.

procount, 2025-02-09:
I have made a quick override, so please proceed as follows:

1. Boot into PINN and wait for the network to connect and the OS list to display.
2. Reboot PINN again and Batocera for RPi3 should be restored.

procount, 2025-02-09:
I have provided a temporary override for Batocera_RPi3 to be located.
Please proceed as follows:

1. Boot into PINN until the network is connected and the list of installable OSes is available
2. Reboot PINN again (unplug/replug power, or press M,Esc,Esc) and Batocera_RPi3 should now be installable.

appscaptain, 2025-02-10:
Thank you! This works perfectly!

closed 2025-02-10 by appscaptain