procount/pinn #880: Moving and extending partitions using GParted breaks PINN systems?

appscaptain, 2025-02-12:
I installed PINN on a 32 GB SD card, added 4 systems (Batocera, Kodi, DietPi and RaspiOS 64 Bit). But I made my DietPI partition only 8 GB. I realized I need more space for DietPI to install Home Assistant. So what I did in Gparted was to reduce the size of some previous partition and then move the DietPI partitions to the left so I could expand it to around 15 GB.

So the order was not changed. Partitions were simply moved to the left and reduced in size.

And I needed to upgrade to Bookwork (because for some reason it seems the DietPI that PINN installs is an outdated Bullseye version), but upon running:
`bash <(curl -sSf ‘https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade’)`

I now get 
```
bash: /dev/fd/63: No such file or directory root
curl: (23) Failure writing output to destination
```

Also I can’t login through SSH anymore - I keep getting wrong password even though the password is correct and works fine locally.

Something broke after the resize, because the exact same upgrade operation and SSH worked perfectly before before I had resized (yes, I have reinstalled DietPI many times :-).
```

df -h
lsblk -o name,fstype,label,size,ro,type,mountpoint,partuuid,uuid
```

This is the output (it's a screenshot because I can't SSH into the system):
https://share.zight.com/o0u1KB0j

I guess this must be related to partitions?

I have tried to completely using Replace from PINN recovery to reset the DietPI installation, but it still causes this error. 

Only DietPI breaks like this. After the resize KODI, RaspiOS and Batocera still work fine (albeit Batocera gives some resize error upon every boot, but it continues booting anyway and works fine after this).

Any ideas?

procount, 2025-02-12:
Yes, I'm afraid moving or resizing partitions with parted/Gparted does seem to break things.
The main reason is that it changes the disk-ID of the drive, which affects all the PARTUUIDS of all the partitions.
This normally only affects OSes on USB drives as they generally use PARTUUIDS.
When on the SD, PINN tends to use devices (e.g. /dev/mmcblk0pX) as the partition reference, so I'm not sure if this is the same cause or whether Gparted has affected something else.

If you can identify the previous Disk-ID prior to using parted, then you can restore it with fdisk and that usually sorts the problem out. (But I would have thought using the Replace function would also have worked...)

An alternative to using parted/gparted is to use PINN to backup each OS to a USB stick, then reinstall the backups and alter the partition size before they are installed through PINN's dialog. But it is rather long winded.

> Also I can’t login through SSH anymore - I keep getting wrong password even though the password is correct and works fine locally.

Do you mean SSH into Dietpi, or PINN?

> Batocera gives some resize error upon every boot, but it continues booting anyway and works fine after this.

Probably some Batocera startup script. Many OSes on a Raspberry Pi do not expect to be multi-booted and believe they have the whole SD card to themselves. I used to remove such scripts because they are unnecessary under PINN, but they hide in many places so are difficult to find and they all tend to fail gracefully, anyway.

appscaptain, 2025-02-12:
Ok, thanks.
1. I don't know what the original ids are as I never thought about it, but it seems like mmcb1k0p3 and mmcb1k0p4 are missing.

2. I mean SSH into dietpi.
3. The batocera script is no issue. I can disable it and it continues booting anyway.

Hmmm... need to think what to do..

appscaptain, 2025-02-12:
Could be awesome with a fix ID feature as part of the fix feature b.t.w. :-) I also tried to run those fixes with no solution.

I imagine the scenario here of resizing partitions could be common especially as we are dealing with SD cards. Maybe whenever PINN reboots it would capture the original disk IDs and then whenever someone has reorganized the partition ID's with Gparted or similar, it could detect this on boot and ask if it should update the disk ID numbers to the correct values.

procount, 2025-02-12:
P3 and P4 are purposefully missing. They are only used by a couple of (deprecated?) OSes.
P2 is an extended partion holding p5-p63 logical partitions that are used for the main OSes.

appscaptain, 2025-02-12:
I did backups of the 3 other systems and wiped the SD card and reinstalled with the 3 backups and DietPi.

Get the same errors as before and can't login.

upon running:
bash <(curl -sSf ‘https://raw.githubusercontent.com/MichaIng/DietPi/master/.meta/dietpi-bookworm-upgrade’)

I now get

bash: /dev/fd/63: No such file or directory root

Also I can’t login through SSH anymore - I keep getting wrong password even though the password is correct and works fine locally.

And I can't log into the system locally either after setting my password...