procount/pinn #882: Manual booting between OSes

Marxon, 2025-02-14:
Hi procount, 

according to the wiki you need to change the "bcmXXXX" part of the command depending on the raspberry modell. 
What to use on a pi4? 
There are serveral different bcm 2835 folder inside /sys/module so i dont know what is the correct one.

Thank you in andvance

procount, 2025-02-14:
In most OSes you can just use `sudo reboot <n>`.
That is far easier. Have you found an OS where that doesn't work?

For the 2nd method I guess it would be bcm2711 for a Pi4.

Marxon, 2025-02-15:
Really that simple? Nice! 
I didn't know about it. 
You should add this info to the wiki.
Issue closed. :)

Thanks a lot and have a nice day!

Marxon, 2025-02-16:
Sorry, I closed the issue before i testing it.  :(
But unfortunately
`reboot <n>`
Boots the first OS everytime. No matter what <n> 

For the 2nd method
`su -c 'echo <n> > /sys/module/bcm2711/parameters/reboot_part'`
 I get 
`can't create /sys/module/bcm2711/parameters/reboot_part: nonexistent directory`

procount, 2025-02-16:
It appears the 2nd method has now been removed from the firmware (or changed in some way) so I shall have to update the wiki.

As you have a Pi4, please see this issue:
https://github.com/procount/pinn/issues/318
and in particular this comment:
https://github.com/procount/pinn/issues/318#issuecomment-633046899
which leads to:
https://forums.raspberrypi.com/viewtopic.php?p=1664625#p1664625

You need to add this argument to the cmdline.txt of each of your installed OSes. You can do that in PINN by going to the Maintenance Menu, selecting each OS in turn and clicking the Edit Config button which allows editing of the config.txt and cmdline.txt file. Add the quirks argument to the first line of cmdline.txt, which should be the one and only line of text.

This is due to a hardware issue of the new PMIC that was added to the early Pi4.

You never mentioned what values of `<n>` you used for the reboot command. This cannot be arbitrary. It represents the partition number of the boot partition of the installed OSes. You can find this from the `installed_os.json` file which is on the settings partition (5) of the PINN drive. If you are in PINN and drop to the command line, this partition is mounted  on `/settings`.  The first installed OS is always on partition 6. The next is probably on partition 8, depending on how many partitions the first OS has.

Marxon, 2025-02-21:
Sorry for my late reply.
Adding `sdhci.debug_squirks2=4` to the cmdline.txt solved the issue and 
`reboot <n>` boots the appropriate OS now. 
Thank you again. :)

closed 2025-02-21 by Marxon