procount/pinn #883: if networktimeout is set and occurs, error dialog: "error downloading distribution list from internet" blocks the boot process

magowiz, 2025-02-15:
---
name: Bug report
about: Create a report to help us fix your issue

---

**Describe the bug**
Add a clear and concise description of what you think the bug is. If I set the option networktimeout to 5 seconds, in order to avoid booting default OS if no network is available (typically on power shortage), PINN behaves well and goes straight (after boot timeouts I set) to picking default OS and boot, but before booting it hangs on the dialog "error downloading distribution list from internet" that blocks the boot process until you click ok.
I'm using PINN in a headless configuration, so I expect the process won't stop and continue straight to default OS even if network is not available at the moment.

**To reproduce**
1. setup pinn with a networktimeout=5
2. save and let it boot to default OS
3. turn off the router to simulate network shortage/absence
4. reboot from default OS
5. see PINN progress on Initialising disappear correctly
6. see the main menu showing for 30s as per timeout and disappearing
7. see the boot menu showing for 5s as per timeout and disappearing
8. a dialog with error "error downloading distribution list from internet" appears and avoid booting process to continue
**Expected behaviour**
After the boot timeout an error about downloading distribution list may be shown or not, but since we had a network timeout, it shouldn't block the booting process

**Actual behaviour**
dialog "error downloading distribution list from internet" that blocks the boot process even if there was a networktimeout.

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? Pi4 Model B
* Which version of PINN? 3.9.3

magowiz, 2025-02-15:
Mine connection for RPi4 is made like this: RPi4 -> MINE-R -> ISP-R -> internet, where *-R are routers, while reporting the bug I was tuning off ISP-R.

I tried also turning off MINE-R and in this case behaved differently and how I expected: I got some no connection errors, wifi dialog, but at the very end, after all timeout I set, the default OS booted without interruptions, so probably PINN detects correctly the connection determining if establish a connection with its router is successful or not, but somehow is not detecting internet reachability.

I think the scenario I wanted to fulfill is working fine, because when I have a power shortage both MINE-R and ISP-R goes down, tell me If you wish I keep it open because you want to deal with reachability or if I should close it.

procount, 2025-02-16:
Could you please post the contents of PINN's cmdline.txt so I can see all the options you have set?

> If I set the option networktimeout to 5 seconds, in order to boot default OS if no network is available

I'm not sure that will fix your issue. It's intention is to delay startup until the network has become available, especially as some wifi router connections take a long time to establish, thus avoiding unnecessary dialog boxes. I normally have to set mine to 30 secs.

I take the point about dialog boxes popping up blocking the boot process if you just want to boot directly to the default OS. I will look into that.

magowiz, 2025-02-16:
yes, sure, here it is my cmdline.txt contents:
`forceupdatepinn repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json forcetrigger remotetimeout=30 networktimeout=5 vncshare ssh bootmenutimeout=5 quiet ramdisk_size=65536 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4
`