procount/pinn #885: PINN OS versions cause errors on bootup

Aszki, 2025-03-01:
Tried on RPi5 16Gb, then 8Gb, same results/

Fresh install of PINN, then LibreELEC and Batocera.

LibreELEC opens to an empty hud: https://www.reddit.com/r/kodi/comments/s3ja5c/libreelec_causes_blank_screen/

Batocera causes this to appear: https://imgur.com/a/2FL2Z9V

When I flashed the newest LibreELEC nightly version to the same SSD by itself, it worked fine. Same with Batocera. Is there a fix for these issues? Tried looking into creating a custom repository with newer versions of these images but was unable to figure out how to format it myself.

Thanks!

procount, 2025-03-01:
I just installed both of these onto USB using an RPI5-8GB rev 1.0 (board revision d04170 - the C0 stepping).
Both worked fine. LibreELEC did a self-update from 12.0.0 to 12.0.2 on second boot and it still worked fine.

The RPi5-16GB version will definitely have the D0 stepping of the processor, so it may have to wait until I upgrade these to a version that supports the D0 stepping. I've heard that the D0 stepping will percolate to all the models eventually, so maybe you also have a D0 stepping in your 8GB version?

Aszki, 2025-03-01:
Hi, thank you very much for replying!

Sorry I only just bought my first raspberry pi the other day so I'm still getting familiar with the terminology.
I returned the 16Gb RPi 5 and bought an 8Gb version which arrived earlier, and still same issue.

After running a quick terminal command it says "Revision: d04171".

Why would this be happening, and is there any fix? Terribly sorry if I didn't provide enough information, I really appreciate your help!

procount, 2025-03-01:
You have a 1.1 revision, but this doesn't indicate if it uses the C0 or D0 stepping, although it could do. You might be able to read the stepping off the top of the processor if you have a keen eye.

Unfortunately, the D0 stepping of the processor requires a later version of the kernel and possibly the firmware too. 
PINN does not support it yet and probably few of the OS versions it installs support it either.

As the OSes support the D0 stepping, and as I have time, I will include the newer versions and endeavour to update PINN to support the D0.

NumberOneGit, 2025-03-28:
I wouldn't say this is exactly the case.  I have been using my CM5 since we discussed this a few months ago.  The eMMC won't work until the kernel is updated, but if the user adds the bcm2712d0-rpi-5-b.dtb as the only dtb in their folder, the board may boot.  The firmware options provided are a little confusing and since I only own my CM5 with d0 and a c1 pi 5 (no pi 5 d0), I can't say with any certainty.   
It IS possible, now that i think of it, that it's an issue not of the hub not properly initializing the emmc, but of the hub itself not initializing.  I hadn't seen posts by other users having issues with PINN on d0 Pis, so I had assumed it was an eMMC thing.  If that's the case, then users can probably use a usb to boot in the interim.  The majority of more popular OS's available on PINN have supported d0 for awhile now.  I have been using PINN on nvme but would be glad to re-test installing it on USB to figure out whether any extra steps were needed to get it installed.