procount/pinn #886: Enabling PINN at boot via GPIO with standard conf files

Ircama, 2025-03-06:
Wondering how to activate PINN at boot via GPIO using the [GPIO Filter](https://www.raspberrypi.com/documentation/computers/config_txt.html#the-gpio-filter) of the operating system CONFIG.TXT file generally located in the /dev/mmcblk0p6 boot partition.

I am aware that PINN has appropriate [cmdline options](https://github.com/procount/pinn/blob/master/README_PINN.md#cmdline-options), but I usually disable it with [autoboot.txt]( https://www.raspberrypi.com/documentation/computers/config_txt.html#autoboot-txt), since I need the fastest boot process in standard mode.  

Before checking the GPIO options, for preliminary testing purposes I attempted to move the config.txt, cmdline.txt, pinn.rfs, and wpa_supplicant.conf.bak files to /dev/mmcblk0p6 in order to verify the possibility to start PINN there. Unfortunately, I was unable to successfully boot it.

Could you please provide some guidance or suggestions on how to achieve this (provided that it is possible)?

procount, 2025-03-06:
I'm a bit confused. Do you really want to "activate PINN at boot via GPIO", or do you actually want to select one of PINN's installed OSes automatically at boot time, based on the status of various GPIO pins, so that you can boot into it as fast as possible by bypassing PINN?

Ircama, 2025-03-06:
I am testing a Raspberry Pi Zero W with a legacy 32-bit OS.

As a default, I would like to directly boot that target OS. Upon a GPIO, I wish to substitute the target OS boot with PINN. I guess that the CONFIG.TXT in the /dev/mmcblk0p6 boot partition might allow this with something like

```ini
...
initramfs initrd.img-5.15.32+-overlay
# end of default settings

[gpio4=1]
cmdline=pinn_cmdline.txt
kernel=pinn_kernel.img
gpu_mem=16
start_file=start_pinn.elf
fixup_file=fixup_pinn.dat
disable_overscan=1
initramfs pinn.rfs
```

procount, 2025-03-06:
Don't move the PINN files - you'll break it, as you found out.

A better solution is just to extend your autoboot.txt file with similar conditionals as above to select the correct boot_partition.
Have a look at https://forums.raspberrypi.com/viewtopic.php?p=2242387&hilit=PINN#p2242387 for an example.
But be aware that if you have multiple OSes installed, that example of consecutive boot_partition numbers is probably inappropriate for the general case.

Ircama, 2025-03-06:
Thank you @procount for clarifying that the idea of moving PINN files is not appropriate.

Extending autoboot.txt would be ideal, but I was initially discouraged of testing it [by the manual](https://www.raspberrypi.com/documentation/computers/config_txt.html#autoboot-txt), which mentions that the supported filters are [all], [none] and [tryboot], so not GPIO apparently.

Instead, your link appears elucidatory. I’ll post the results of my testing.

I do not have multiple OS installed and I’d love to preserve the fast boot of using autoboot.txt, while opening the possibility to run PINN through a GPIO.

Ircama, 2025-03-07:
It works. This autoboot.txt for example allows booting either PINN or the OS depending on the value of the GPIO:

```ini
[gpio21=1]
boot_partition=6
[gpio21=0]
boot_partition=1
```

Thanks again.

closed 2025-03-07 by Ircama