procount/pinn #891: Need to choose wifi on every boot

jeffnwarner, 2025-04-01:
---
name: Bug report
about: Create a report to help us fix your issue

---

**Describe the bug**
Add a clear and concise description of what you think the bug is.
In the README it says "PINN will remember your wifi network connections, so it is only necessary to enter them once" 
However, every time I boot up my Raspberry Pi 5 without an ethernet cable it asks me to choose my wifi to connect to. 

**To reproduce**
List the steps required to reproduce the issue.
Boot up Raspberry Pi 5 without ethernet 
Connect to wifi
Reboot Raspberry Pi 5
Wifi will be asked for again

**Expected behaviour**
Add a clear and concise description of what you expected to happen.
PINN will remember your wifi network connections, so it is only necessary to enter them once

**Actual behaviour**
Add a clear and concise description of what actually happened.
I am being asked every time

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 5

* Which version of PINN?
3.9.3

**Logs**
If applicable, add the relevant output from `/tmp/debug` or `dmesg`.

**Additional context**
Add any other relevant context for the problem.

procount, 2025-04-01:
After you have entered your wifi details and connected to your wifi and rebooted, I guess it pops up with a dialog with the title "No network access". Instead of pressing the "close" button that leads to the wifi settings dialog, what happens if you leave it for a while (up to one minute)? Do you see the list of OSes appear in the background and an IP address appear in the menu title bar?

I suspect your wifi is taking longer to connect than PINN expects so it is timing out. If so, go to the maintenance menu. PINN should already be highlighted, so just click on the "Edit Config" button. Then click on the "Edit Options" button. Select the "Booting" tab and enter a longer timeout value in the networktimeout box. I suggest trying 30 - that usually works for me, but make it longer if it still keeps occurring.

jeffnwarner, 2025-04-01:
Thanks that fixed the issue!

closed 2025-04-01 by jeffnwarner