procount/pinn #892: LibreELEC boot error via PINN

starlord-87, 2025-04-01:
Hi,

I'm a noob here and sorry if this has been posted somewhere already but I wasn't able to find anything in my search for a fix, so feel free to point me in the right direction.

I was trying to setup a multiboot via a confirmed usable NVMe SSD on a Pi 5 but I keep running into the following error on trying to boot into LibreELEC to even set it up.

[image]

My current setup for the multiboot is as follows - using a Raspberry Pi 5 16GB inside a Pironman 5 Case and using the M.2 HAT to boot from a Cruical P3 Pro SSD. Current OSes installed on the SSD for the multiboot setup are full Pi OS 64-bit, Android TV via Konstakang's Lineage OS 22 build and LibreELEC. I also, included a data portion during the initial PINN setup.

Was able to get Pi OS up and running without any issues, but I had to add-in dtoverlays to the config.txt file to get Lineage OS to boot. I'm wondering if I need to do the same for LibreELEC but I couldn't find what dtoverlays would be needed to get the boot into the OS to work.

P.S. Everything on the Pi 5 is up-to-date, did all the firmware and eeprom updates to use the latest bootloader but still getting the issue booting into LibreELEC OS.

procount, 2025-04-01:
As you have a 16GB Pi5, that means you are using the D0 stepping of the processor which requires the very latest firmware/kernel. I suspect LibreELEC does not contain these, hence the fatal firmware error. You could try updating these yourself in LibreELEC, but I've no idea if that would work, or wait until there is a new LIbreELEC version on PINN that works with the D0 stepping.

starlord-87, 2025-04-02:
@procount I tried a clean only boot of LibreELEC using Pi's imaging tool on a SD card which booted into LibreELEC using the same 16GB Pi5.

Is it possible to then copy the firmware/kernel file from the SD card to the correct boot partition PINN created for LibreELEC and see if that works?

P.S. My first time really playing around with a Pi, so will need some guidance to figure out which files would need to be copied to where, if this is possible.