procount/pinn #893: Update from 3.9.2 fails due to “Partition size too small”

hupfdule, 2025-04-07:
I have exactly the same problem as https://github.com/procount/pinn/issues/817, but in my case I currently have PINN 3.9.2.

The upgrade fails silently. In /tmp/debug log I can see:

```
+ sync
+ [ 63950 -lt 64076 ]
+ echo Partition size Too small
Partition size Too small
+ exit 1
" 
 
+ error=1
+ [ x1 == x ]
+ output=/tmp/mnt
+ [ ! 1 -eq 0 ]
+ cp -a /tmp/mnt/BUILD-DATA /tmp/mnt/INSTRUCTIONS-README.txt /tmp/mnt/Microntek______________USB_Joystick___________6A_12B.json /tmp/mnt/Microsoft_X-Box_360_pad_8A_11B.json /tmp/mnt/Microsoft_X-Box_pad_v1__US__8A_10B.json /tmp/mnt/RECOVERY_FILES_DO_NOT_EDIT /tmp/mnt/Sony_PLAYSTATION_R_3_Controller_27A_19B.json /tmp/mnt/Sony_PLAYSTATION_R_3_Controller_30A_13B.json /tmp/mnt/Sony_PLAYSTATION_R_3_Controller_7A_13B.json /tmp/mnt/Xin-Mo_Xin-Mo_Dual_Arcade_2A_22B.json /tmp/mnt/bcm2708-rpi-b-plus.dtb /tmp/mnt/bcm2708-rpi-b-rev1.dtb /tmp/mnt/bcm2708-rpi-b.dtb /tmp/mnt/bcm2708-rpi-cm.dtb /tmp/mnt/bcm2708-rpi-zero-w.dtb /tmp/mnt/bcm2708-rpi-zero.dtb /tmp/mnt/bcm2709-rpi-2-b.dtb /tmp/mnt/bcm2710-rpi-2-b.dtb /tmp/mnt/bcm2710-rpi-3-b-plus.dtb /tmp/mnt/bcm2710-rpi-3-b.dtb /tmp/mnt/bcm2710-rpi-cm3.dtb /tmp/mnt/bcm2710-rpi-zero-2-w.dtb /tmp/mnt/bcm2710-rpi-zero-2.dtb /tmp/mnt/bcm2711-rpi-4-b.dtb /tmp/mnt/bcm2711-rpi-400.dtb /tmp/mnt/bcm2711-rpi-cm4.dtb /tmp/mnt/bcm2711-rpi-cm4s.dtb /tmp/mnt/bcm2712-rpi-5-b.dtb /tmp/mnt/bootcode.bin /tmp/mnt/cec_keys.def /tmp/mnt/cmdline.bak /tmp/mnt/cmdline.txt /tmp/mnt/cmdline.txt.ORIG /tmp/mnt/config.bak /tmp/mnt/config.txt /tmp/mnt/defaults /tmp/mnt/firmware.tar.gz /tmp/mnt/fixup.dat /tmp/mnt/fixup4.dat /tmp/mnt/joy_keys.def /tmp/mnt/kernel.img /tmp/mnt/kernel7.img /tmp/mnt/kernel7l.img /tmp/mnt/kernel8.img /tmp/mnt/overlays /tmp/mnt/overlays6 /tmp/mnt/overrides.json /tmp/mnt/pinn.rfs /tmp/mnt/pinn_init /tmp/mnt/postupdate /tmp/mnt/preupdate /tmp/mnt/riscos-boot.bin /tmp/mnt/screenpref /tmp/mnt/start.elf /tmp/mnt/start4.elf /tmp/mnt/wallpaper.jpg /mnt
+ exit 1
update failed. 
rm: can't remove '/settings/BUILD-DATA': No such file or directory
```

procount, 2025-04-08:
> The upgrade fails silently.

Did it rollback to 3.9.2 or is it non-functional now?
Unless you have one of the latest D0 steppings, I suggest sticking with the 3.9.2 version.
To upgrade any further you really need a bigger PINN partition, so it's best to start afresh. See https://forums.raspberrypi.com/viewtopic.php?p=2220339#p2220339 for more details on your options, although using Gparted can mess up your disk IDs and make some of your OSes non-bootable without some manual intervention.

hupfdule, 2025-04-10:
It is still functional. There was no rollback. It seems that installation just failed, but didn’t break anything.
So I will just stay on 3.9.2. Thanks for the hints.