procount/pinn #895: How would I install 2 copies of an OS that wants 3 partitions each (2x recalbox)

Gremious, 2025-05-01:
I want to dualboot 2 copies of recalbox, so that me and my partner can separate our configs/games/saves/retroachievements login/etc.

Project spaces give you a 2 partition space, and that OS wants 3 (settings, recalbox, shared), so while I can install one copy + a project space, I cannot install a second copy as the partition count mismatches.

What is the best way to achieve this?

On that note, perhaps project spaces for 3 (and up?) partitions would be a good feature.

procount, 2025-05-01:
A ProjectSpace is just a dummy OS comprising a few meta files that describe the required partition space so that it may be allocated by PINN to reserve some space. 
The standard projectspace is fine for the majority of "standard" OSes, like Raspios or Ubuntu, but if you have something other than this, then you need to create your own projectspace.

To create your own projectspace, I suggest something like the following (untested):
Take a USB memory stick (preferably formatted as FAT32, but ext4 is also acceptable).
Create a folder on it called 'os' in the root of the drive.
Plug your memory stick into your Pi with PINN running. The new drive should show up in the destination drive box on the Main Menu and Archive Menu screens. 
Press the Clear button to ensure no OSes are selected, then select ProjectSpace1 and choose to download it from the Archive Menu.
Look at the files in the /os folder. Change the projectspace name to something else (e.g. recalbox_projectspace). You can test it by rebooting PINN to reload and check it appears in the utility tab.
Now modify the partitions.json file to show 3 partitions. Use the recalbox partitions.json file as a template for what it should look like.
You should now be able to install recalbox and the recalboxproject space to your Pi, and replace the project space with a second copy of recalbox.

The above will allow you to install 2 copies of recalbox, but there is another factor to consider.
Some OSes require partitions to have certain names, and some upgrade procedures may fail if they are named differently.
PINN will adapt partition names and their references to make them unique, so the first OS will have the correct names, but the second OS will have a '0' suffix on each partition name. However, if say, an upgrade process on the second OS references SHARE instead of the adapted SHARE0 name, it may affect the data of the first OS instead of the second OS. 

I can't remember if recalbox _requires_ certain partition names, so you're best to just try it and see. Just be aware of this limitation. Most OSes assume they will be the only OS installed and are not aware of multi-boot environments.

Gremious, 2025-05-01:
That did work and led me to 2 instances of recalbox on boot - so thank you very much!

Unfortunately, it seems that while it makes 2 boot partitions, it only makes 1 shared SETTINGS partition, and then the share space is up to you (and iirc you can change SHARE in the settings anyway for external drives.)

[image]

So I believe my best bet would be to just have two sd cards/usb sticks with separate recalbox installs that we just keep around to plugin whenever we play.

Oh well. Thank you very much for your time :)

---

For anyone else's reference, my project space partition jsons looked something like this.

Also, i did not realize initially, but I had to put said custom project space "`os`" on the same stick as PINN, does not work on a separate usb. 

<details>
  <summary>json ⬇️</summary>
  

```json 
{
 "partitions" : [
  {
   "filesystem_type" : "FAT",
   "label" : "prjboot",
   "partition_size_nominal" : 3071,
   "mkfs_options" : "-F 32",
   "tarball" : "http://downloads.sourceforge.net/project/pinn/os/ProjectSpace/prjboot.tar.xz",
   "uncompressed_tarball_size" : 1,
   "want_maximised" : false
  },
  {
   "empty_fs" : true,
   "filesystem_type" : "unformatted",
   "label" : "SHARE2",
   "partition_size_nominal" : 3072,
   "partition_type" : "07",
   "uncompressed_tarball_size" : 0,
   "want_maximised" : true
  },
  {
   "filesystem_type" : "ext4",
   "label" : "prjroot",
   "mkfs_options" : "-O ^huge_file",
   "partition_size_nominal" : 1024,
   "tarball" : "http://downloads.sourceforge.net/project/pinn/os/ProjectSpace/prjroot.tar.xz",
   "uncompressed_tarball_size" : 1,
   "want_maximised" : false
  }
 ]
}
```

</details>

closed 2025-05-01 by Gremious