procount/pinn #897: Dietpi_64 needs update

Lonerider2010, 2025-05-01:
See [Dietpi_64 needs update #28](https://github.com/matthuisman/pinn-os/issues/28).
Just wanted to make sure to know when I can use the update, especially with rpi-installer. Thanks.

procount, 2025-05-02:
I've copied DietPi_RPi5 from Matt's repository, so it should now be available.

Lonerider2010, 2025-05-02:
@procount just wanted to start testing, but "calling home" is currenty not possible, see [DietPi maintenance](https://dietpi.com/maintenance.html)

Andymc1280, 2026-04-06:
Trying to use Dietpi x64 with pinn on a CM5, but the 9.2 version wont finish install process/update.  Any chance of a newer 10.x based image ?

procount, 2026-04-27:
Done!
I included basic configuration for the Rpi5 version to work on the OneUp CM5 laptop too.