procount/pinn #902: DSI Support for Raspberry Pi Touch Display 2

Eclipse00, 2025-05-23:
I have tried many configurations in the cmdline.txt and in config.txt and I can't get the Raspberry Pi Touch Display 2 to post with PINN. Not sure if I'm missing anything, or there isn't support yet for it. Is this a known compatibility issue, if not is there a less then obvious resource or guide for making it work.

Eclipse00, 2025-05-23:
Apologies for not providing all relevant data upfront, I'm using an RPI4 and it does work in raspbian and Ubuntu without issue. I'll try as suggested and let you know if I get results. Thanks for taking the time to respond!

procount, 2025-05-23:
The one thing that might stop it working is if it needs any drivers that PINN does not include. PINN only includes a very minimal set to keep it small.