procount/pinn #904: RPi5 8Gb With NVME Boot some OS Error

ronnytiaka, 2025-05-29:
**Describe the bug**
RaspiOS 64 Full - frezee at splashscreen
RapsiOS Lite - Error Can't have a partition outside the disk!
Ubuntu - needs a partition label of 'system-boot' which is not available

**To reproduce**
Install UBUNTU Desktop 25.04 64Bit using Cardreader to MicroSD using Rpi Imager with windows 11
Put the microSD to RPi5 8Gb
Put NVME Hat to RPi5 
Install RPi Imager on Ubuntu Desktop
Install PINN to NMVE
boot NVME on RPi5
Install OS RaspiOS 64 Full, RapsiOS Lite, Ubuntu24.10, kali64, Recalbox, LibreElec, Swap using Wifi

**Expected behaviour**
kali64, Recalbox, LibreElec is Succeed to boot

**Actual behaviour**
But Some os Error :
RaspiOS 64 Full - frezee at splashscreen
RapsiOS Lite - Error Can't have a partition outside the disk!
Ubuntu - needs a partition label of 'system-boot' which is not available

**System**
RPi5 8Gb
NMVE Micron 2230 512Gb M.2 Gen4

procount, 2025-05-29:
That's quite odd. I have often & recently installed all of the RaspiOS installations without issue.
Ubuntu _requires_ a partition label of 'system-boot' in order for upgrades to work properly. This prevents 2 instances of Ubuntu from being installed. The error suggests that this partition label has already been taken by one of the other OSes, so maybe its just the combination of OSes that you have installed?

I will try to replicate your issue.
I am about to release new installations of RaspiOS following the recent announcement, so maybe that will also help?

procount, 2025-05-29:
I have tried to replicate your procedure on a RPi5 8GB with SSD. Not exactly, because I used Netinstall rather than using Ubuntu 25.04 with RPi-Imager to install PINN. But I think maybe that is why PINN could not Ubuntu 24.10, because you still had Ubuntu 25.04 on the SD card plugged in, which would already have a system-boot partition. You can't have 2 partitions with the same name at the same time, so please remove your SD card before installing Ubuntu24.10.

It took a while to download & install all of those OSes, so I left it unattended. They all seemed to install successfully, although when I reviewed the logs, I saw there was an "Error Can't have a partition outside the disk!" message, but I couldn't attribute it to any particular OS as it occurred just after the disk was partitioned. I'm not sure this is an issue.

However, neither RaspiOS would boot up properly from nvme, not even with the latest release. I think something in RaspiOS must have changed, but I will look into this.

ronnytiaka, 2025-05-30:
Yes with Wifi is take time to install, i take more than 12 Hours 🤭. so i now i try using "Local Images on SD card" method, like this step :

1. Im Running Ubuntu 25.04 Desktop on SDCard
2. Delete All Partition on NVME using Disk
3. Format and Create 25GB partition with FAT32 Using Gparted
4. Download PINN_lite.zip dan Extract to Fat32 Partition
5. Download OS from https://sourceforge.net/projects/pinn/files, kali64_5,kde5slee64, LIbreElec_RPI5, raspiod_arm64_lite, recalbox5 and raspios_arm64
6. i put the os in folder "os" in Fat32 Partition
7. i remove SDCard Ubuntu 25.04
8. boot to NVME, change space partition each OS and install

It still same won't boot RaspiOS either, 
Error : Can't have a partition outside the disk!

procount, 2025-05-30:
Could you please do the following checks for me?
Please boot into your Ubuntu SD card and in a terminal capture the output from `sudo fdisk -l`. Please post it here with the contents of `installed_os.json` from the `settings` partition (it should be the partition `/dev/nvme0n1p5` which may already be mounted)

ronnytiaka, 2025-05-31:
I Try  using "Local Images on SD card" method

This my partition setup :

[image]

Ubuntu Works, with SDCard remove before boot to NVME and installing PINN OS

[image]

But RaspiOS still frezee !!!

**this Disk Screenshut :**

[image]

fdiskscreenshut :

[image]

fdisk Copy to text :

[fdisk.txt](https://github.com/user-attachments/files/20530323/fdisk.txt)

installed_os.json on **settings partition**

[installed_os.json](https://github.com/user-attachments/files/20530324/installed_os.json)

procount, 2025-06-06:
The problem with Raspios versions freezing on SSD has been identified.
RPL have been informed and I expect it will be fixed in their next release.
In the meantime, I am temporarily fixing the converted files and will post them once complete.

ronnytiaka, 2025-06-07:
Thanks, when do you plan to release the PENN update? Or you just fix raspios zip os

ronnytiaka, 2025-06-09:
Thanks 🙏👌😁, current status is :
RapsiOS Lite - **Success**
Ubuntu24.10  - **Success**
kali64 - **Success**
kde5lee64 - **Success**
Recalbox - **Success**
LibreElec - **Success**

RaspiOS 64 this detail :
**Success Boot**, **but** when conencting to wifi in installing process is frezee or make input output error
![Image](https://github.com/user-attachments/assets/b2ea2f1d-4f39-4b47-a619-aeec9731caed
I need to unplug the power repeatedly till wifi is connected, i dont know is because your OS zip, or maybe is problem from RaspiOS itself. 
if this problem is from RaspiOS itself, i thinks is bugs is Closed, but if this because you OS Zip compilation, i wait for progress