procount/pinn #906: Wifi interface is not available

maxy-maker, 2025-06-05:
---
name: Bug report
about: Create a report to help us fix your issue

---

**Describe the bug**
Today I installed pinn 3.9.3 using the rpi imager on my rpi 5 16gb ram and for some reason it didn't see the wifi interface. No mater how many restarts I did, it never worked and gives me empty window.

**To reproduce**
Installing pinn from rpi imager on rpi 5

**Expected behaviour**
 wifi ssids to choose from the wifi

**Actual behaviour**
Blank and gives me error msg

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* pi5 16gb ram
* Which version of PINN?
3.9.3 from imager
**Logs**
If applicable, add the relevant output from `/tmp/debug` or `dmesg`.

**Additional context**
Add any other relevant context for the problem.

procount, 2025-06-05:
Unfortunately, I don't have a Pi5 16GB version to test (only 8GB version), but v3.9.3 is suitable for the D0 stepping of the processor and I haven't had any previous reported issues on the wifi not working with it, so I assume this is not a problem with PINN.

Have you tried installing plain Raspios without PINN? Does that work with wifi?

Some wifi connections can take a while to establish.  Please try pressing the More button on the menu twice to get to the Maintenance Menu. Click the Edit Config button. Click the Edit Options button. Select the booting Tab. Enter a networktimeout of 60. Click OK twice to return to the menu. Press Esc twice to reboot. See if that helps.

What country are you in?

maxy-maker, 2025-06-05:
Initially, I suspected a hardware fault; however, after a successful Raspberry Pi OS installation via Ethernet using PiNN, the Wi-Fi functionality was confirmed operational.  But, Wi-Fi access remains unavailable when connecting through PiNN.

maxy-maker, 2025-06-06:
Hello again procount, I actually tried the 60-second network timeout, but it didn't fix it, although the Wi-Fi is working well with Raspberry Pi OS that is installed via PINN. Any other suggestions? Thanks

procount, 2025-06-06:
Hmm.

Pinging @XECDesign - is there any difference between the C and D steppings (8GB & 16GB PI5s) that might require new wifi drivers in PINN? I don't remember any other reports about PINN's wifi not working on the 16GB versions.

closed 2025-06-06 by maxy-maker

XECDesign, 2025-06-09:
> is there any difference between the C and D steppings (8GB & 16GB PI5s) that might require new wifi drivers in PINN? I don't remember any other reports about PINN's wifi not working on the 16GB versions.

Sorry, off the top of my head, I have no idea and I don't have the appropriate boards to check. If your defconfigs and firmware files track ours, it shouldn't be an issue.