procount/pinn #907: Silentinstall broken by adjustSizes dialog

MucusMerchant, 2025-06-05:
I am trying to use PINN to headlessly install an OS without any GUI interaction required. I followed the guide for automatic installation in the README ([https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-automatically-install-oses](here)). I placed the full raspios_bullseye_armhf_lite in a folder inside the /os directory. 

On boot, the OS that I downloaded appears properly in the list. It also seems like the silentinstall option successfully "clicked" install for me, because the adjustSizes dialog ("Adjust size of OSes") for my OS pops up. This dialog, however, prevents the installation from proceeding headlessly. The dialog executes at line 3186 of mainwindow.cpp, regardless of whether silentinstall is specified.

procount, 2025-06-05:
Oops!
Thanks for spotting that. Guess I missed it. I will fix it.

fthiebolt, 2025-07-24:
It works :)

For the sake of completeness, here our 'os_list.json'.
Please, pay attention to the added 'Pi 5' in the 'supported_models'

Enjoy !

{
    "os_list": [
        {
            "description": "myORG custom Raspbian",
            "feature_level": 35120124,
            "icon": "http://xxx/images/Raspbian-IRIT-latest/Raspbian.png",
            "nominal_size": 8192,
            "os_info": "http://xxx/images/Raspbian-IRIT-latest/os.json",
            "os_name": "Raspbian-IRIT",
            "partition_setup": "http://xxx/images/Raspbian-IRIT-latest/partition_setup.sh",
            "partitions_info": "http://xxx/images/Raspbian-IRIT-latest/partitions.json",
            "release_date": "2025-01-13",
            "supported_hex_revisions": "2,3,4,5,6,7,8,9,d,e,f,10,11,12,14,19,1040,1041,0092,0093,2082",
            "supported_models": [
                "Pi Model",
                "Pi 2",
                "Pi Zero",
                "Pi 3",
                "Pi Compute Module 3",
                "Pi 4",
                "Pi 5"
            ],
            "tarballs": [
                "http://xxx/images/Raspbian-IRIT-latest/firmware.tar.xz",
                "http://xxx/images/Raspbian-IRIT-latest/root.tar.xz"
            ]
        }
    ]
}