procount/pinn #910: Dual boot from NVMe: getting to (initramfs) prompt

AmsteinGraphics, 2025-06-09:
See forum RaspberryPi forums post https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2200#p2314495

Attached ar the meta files I've adapted.

[pinn_meta_files.zip](https://github.com/user-attachments/files/20659614/pinn_meta_files.zip)

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2301266#p2301266)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2301266#p2301266) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sat Mar 08, 2025 1:00 pm

Hello, I'm trying to install a dual boot RaspberryOS/RetroPie on my Pi 5 NVMe drive. I have copied the gz compressed RetroPie image to the /os/ folder in a subfolder called retropie.

Of course the OS does not appear in the PINN installation list yet.

What are suporting JSON files and how to find/create them?

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2301480#p2301480)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2301480#p2301480) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sun Mar 09, 2025 3:38 pm

Thank you procount.

Now another problem which I can't seem to solve (I know almost nothing about Linux systems). Whichever OS I install using PINN boots to a command line with prompt "(initramfs)". I've tried RaspberryOS 64-bit Full and RaspberryOS Lite.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2301516#p2301516)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2301516#p2301516) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sun Mar 09, 2025 5:58 pm

Yes, I copied the PINN files to a freshly formatted NVMe drive. Then I booted from there and proceeded to make partitions and install OSes from the PINN GUI.

The OS splash screen shows, but then I end up on the initramfs prompt.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2303718#p2303718)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2303718#p2303718) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Tue Mar 18, 2025 2:56 pm

[procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876) wrote: Tue Mar 11, 2025 12:14 am
If you are still booting into an (initramfs), I suggest you try to sort that out before even attempting to create your own multi-boot installation.
Have you tried booting Raspberry Pi OS on your NVME directly without PINN?
Imaged RaspberryOS 64-bit directly to NVMe. Boots fine. Stay tuned... (Sorry progress is very slow, I'm kind of hyper ultra over busy these days.)

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2305750#p2305750)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2305750#p2305750) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Wed Mar 26, 2025 5:49 pm

I'm trying to make a copy of the RetroPie install I have on an SD card. Issuing command
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

sudo bsdtar --numeric-owner --format gnutar -cpvf ~/os/retropie/boot.tar .
returns
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

bsdtar: Failed to open '/home/emy/os/retropie/boot.tar'
Both partitions from the SD card are mounted and readable.

I'm doing this from RPiOS, and I've installed package bsdtar. What am I doing wrong?

Thanks

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2305904#p2305904)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2305904#p2305904) by [procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876) » Thu Mar 27, 2025 1:53 pm

[Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) wrote: Wed Mar 26, 2025 5:49 pm
I'm trying to make a copy of the RetroPie install I have on an SD card. Issuing command
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

sudo bsdtar --numeric-owner --format gnutar -cpvf ~/os/retropie/boot.tar .
returns
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

bsdtar: Failed to open '/home/emy/os/retropie/boot.tar'
Both partitions from the SD card are mounted and readable.

I'm doing this from RPiOS, and I've installed package bsdtar. What am I doing wrong?

Thanks
1. Did you install bsdtar from APT? If so, best to uninstall it.
You should use bsdtar v3.3.1, which can be installed following the instructions here -> [https://github.com/procount/pinn/blob/m ... pendencies](https://github.com/procount/pinn/blob/master/README_PINN.md#get-build-dependencies)

2. Does the target folder path '/home/emy/os/retropie/' exist?

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2307136#p2307136)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2307136#p2307136) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Tue Apr 01, 2025 8:43 pm

Still working to make 2 custom OSes available to PINN as per your doc here: [https://github.com/procount/pinn/wiki/H ... using-PINN](https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN)

Thanks, building bsdtar per your instructions worked. I'm down to the step where I should output the size of the tar archives to text files (apparently), but command
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

sudo du -BK -s /media/<username>/boot | cut -d"K" -f1 >boot.size
returns
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

bash: boot.size: Permission denied
Why?

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2307138#p2307138)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2307138#p2307138) by [procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876) » Tue Apr 01, 2025 8:58 pm

It means you don't have permission to write boot.size to your current directory, but you didn't say where that was.
Maybe do
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2175#)

cd ~
to put you in your home directory then repeat the command.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2309485#p2309485)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2309485#p2309485) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sun Apr 13, 2025 4:30 am

Hello, thanks for the help.

I'm back with the next step.

I've followed instructions, modified os.json and partitions.json, and finally copied files for both OSes on an USB stick in folder /os. But they still don't appear on the PINN UI (no ethernet, wi-fi disabled). Not even sure where they should appear. The intuition would suggest that there should be some button to direct PINN to search for OSes on local drives, but I understand this isn't necesary when all goes well... :)

I removed checksums from os.json and partitions.json. I'm using an ext4-formatted USB drive and PINN v3.9.3.

Not sure what to try next.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2312137#p2312137)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2312137#p2312137) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sun Apr 27, 2025 2:09 pm

First check that your USB stick is being recognised by PINN.
It is. It appears as an install destination in the drop-down at the bottom of the window.
Try validating your syntax is correct by posting it on the jsonlint.com webpage.
The page validates the syntax.

I'm sorry, I do not know what to do next.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2312365#p2312365)
Post by procount » Mon Apr 28, 2025 4:11 pm

Eamoex wrote: Sun Apr 27, 2025 2:09 pm
I'm sorry, I do not know what to do next.
Try enabling the 'showall' option in the Maintenance Menu. (select PINN, [Edit](https://forums.raspberrypi.com/posting.php?mode=edit&p=2312365) Config, Edit Options, Source tab I think?) and reboot PINN.

Failing that, provide a list of the files on your USB stick. I presume they are in /os/Retropie2/ folder, or similar.
And post the contents of os.json.
You might find it easier to open an issue on my github ( https://github.com/[procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876)/pinn/issues ) if you want to post files or photos.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2314495#p2314495)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2314495#p2314495) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Sat May 10, 2025 2:24 pm

[procount](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=74876) wrote: Mon Apr 28, 2025 4:11 pm
Try enabling the 'showall' option in the Maintenance Menu. (select PINN, Edit Config, Edit Options, Source tab I think?) and reboot PINN.
That did the trick, thanks! After correcting my wrong partition size values and names in partitions.json, and ignoring PINN's warning that none of the OSes were supposedly compatible with Raspberry Pi 5, PINN did install both OSes.

But still no joy. For the first boot I selected the Raspberry OS partition and ended up on an
Code: [Select all](https://forums.raspberrypi.com/viewtopic.php?t=142574&start=2200#)

(initramfs)
prompt... :(

*sigh*

Any idea what could be wrong here?

EDIT: please note, I'm targeting the NVMe drive attached to the PCI port. All installed OS files and partitions on the NVMe drive show up and are readable from a Raspberry OS booted from the SD card slot.

EDIT 2: also, between my first initrams woes and this, I had successfully installed and booted a regular (non-PINN) image of Raspberry OS on the NVMe drive.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2314674#p2314674)
Post by procount » Sun May 11, 2025 11:36 pm

Eamoex wrote: Sat May 10, 2025 2:24 pm
procount wrote: Mon Apr 28, 2025 4:11 pm
Try enabling the 'showall' option in the Maintenance Menu. (select PINN, [Edit](https://forums.raspberrypi.com/posting.php?mode=edit&p=2314674) Config, Edit Options, Source tab I think?) and reboot PINN.
That did the trick, thanks! After correcting my wrong partition size values and names in partitions.json, and ignoring PINN's warning that none of the OSes were supposedly compatible with Raspberry Pi 5, PINN did install both OSes.

But still no joy. For the first boot I selected the Raspberry OS partition and ended up on an
Code: Select all

(initramfs)
prompt... :(

*sigh*

Any idea what could be wrong here?

EDIT: please note, I'm targeting the NVMe drive attached to the PCI port. All installed OS files and partitions on the NVMe drive show up and are readable from a Raspberry OS booted from the SD card slot.

EDIT 2: also, between my first initrams woes and this, I had successfully installed and booted a regular (non-PINN) image of Raspberry OS on the NVMe drive.
If the `showall` option displays your OS, then you haven't included the Pi5 model in the list of suitable models this OS is suitable for in the os.json and os_list.json files.

If you end up with an `(initramfs)` prompt, then there's probably something wrong with the cmdline.txt modifications from partition_setup.sh. Maybe you didn't adapt it for bookworm(?) by removing the space after boot?

If you're "targeting the NVME drive", do you still have PINN on an SD card? You should really have PINN on the NVME drive as well. PINN is not just an installer, but a boot selector, and so it needs to be present on the boot drive. Keeping it on the SD card means you must always have the SD card fitted to boot from.

I'm groping in the dark a bit, because I can't see any of your modified files. Converting an OS for use with PINN is a bit of an art since each one is unique and the conversion may even have to change for a new release.
If you're still stuck, try opening an issue on my github and post your meta files there. It's a bit difficult to follow all the posts in this thread over a long period of time, so keeping them together in a github issue would be easier to manage.

procount, 2025-06-09:
[Re: STICKY: PINN - An enhanced version of NOOBS.](https://forums.raspberrypi.com/viewtopic.php?p=2314698#p2314698)
[Post](https://forums.raspberrypi.com/viewtopic.php?p=2314698#p2314698) by [Eamoex](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=281266) » Mon May 12, 2025 7:16 am

First, thanks for all your help and replies.

The SD card has a separate OS, nothing to do with PINN. Yes, PINN is on the NVMe as well.

I'll "move this topic" to GitHub and post the meta files there.

procount, 2025-06-22:
Recalbox 9.2.3 is already supported in PINN. Why not just install it?

AmsteinGraphics, 2025-06-22:
Oh. Is it? [facepalm]
Going to do that right away.

AmsteinGraphics, 2025-06-23:
So much for nothing.Thanks for your time of advice and development effort. PINN is excellent and makes my dream Pi system come true. Cheers