procount/pinn #915: PINN won't back up custom OS

rickmoon1, 2025-06-29:
Dear Procount

I have successfully made a custom OS for PINN, based on Raspberry Pi OS Bullseye. It is in fact the vanilla image of PiSDR v6.1, which contains many software defined radio applications. The build was done on a Raspberry Pi 5 machine, using Raspberry Pi OS Bookworm 2025-05-08. The PINN/OS test was perforformed on an early Raspberry Pi 4 machine which the original PiSDR OS was built for.

I have successfully installed it using PINN V3.9.2, but when I tried to make a backup, PINN flagged the OS as "unsupported" and it won't perform the backup.

The OS has the 2 partitions - "boot" and "rootfs". I made tar.xz files as directed in your custom os instructions. The tar.xz files have the same names IE boot.tar.xz and rootfs.tar.xz, and I used what I judged to be the nearest corresponding partition.json, os.json and partition_setup.sh files and modified them according to your instructions. The folder that sits in the /os folder was renamed to PiSDR6_64, and the corresponding change was made to to the os.json file to reflect this.

The only issues I encountered when making the rootfs.tar file was a "permission denied" message when I ran the PINN ACL query, and "Cannot set the file group: Operation not permitted" when the tar.xz commands had completed.

Is this a case of renaming the OS to something that is listed in PINN  - eg raspios_bullseye_arm64 ?

I'll attach the CLI output when making the tar.xz files, the PiSDR OS info and the json and sh files I used. Apologies if the CLI file has a few extraneous bits and pieces. This reveals my lack of experience with it !

I hope this makes sense and it's something obvious

Many thanks, 

Rickmoon

[os.json](https://github.com/user-attachments/files/20969199/os.json)
[partitions.json](https://github.com/user-attachments/files/20969198/partitions.json)
[partitions_new.json](https://github.com/user-attachments/files/20969200/partitions_new.json)
[PiSDR6_64 OS Info.txt](https://github.com/user-attachments/files/20969214/PiSDR6_64.OS.Info.txt)
[PiSDR PINN Build Logging 280625.txt](https://github.com/user-attachments/files/20969215/PiSDR.PINN.Build.Logging.280625.txt)

rickmoon1, 2025-06-29:
Hi Procount

Please ignore uploaded partitions_new.json. I should have uploaded partition_setup.sh which I will do now. 

Ah - it won't let me with that extension, so I've changed the file extension to .txt

Many thanks

Rickmoon

[partition_setup.txt](https://github.com/user-attachments/files/20969492/partition_setup.txt)

rickmoon1, 2025-06-29:
Hi Procount

Thanks for your reply. I will add the line to os.json. I forgot to alter the Kernel version as well. 

Rootfs is on an ext4 partition. boot is on a fat32 partition

Many thanks

Rickmoon

procount, 2025-06-30:
> cp: can't stat '/settings/os/PiSDR6_64/icon.png': No such file or directory
> cp: can't stat '/settings/os/PiSDR6_64/release_notes.txt': No such file or directory

It looks like you are missing these 2 files. 
icon.png should be a 40x40 pixel icon to represent your custom OS (it's normally named after your OS name).
release_notes.txt normally describes the latest features of your custom os. Just a short message will do so that it exists.

closed 2025-07-03 by procount