procount/pinn #918: P3.9.7 doesn't work on D0 stepping CPUs

procount, 2025-07-20:


procount, 2025-07-20:
@fprietog - Could you please try 3.9.5 from https://sourceforge.net/projects/pinn/files/archive/pinn-395.zip/download

procount, 2025-07-21:
> Include bcm2712d0.dtbo in overlays folder so that D0 has the correct pinctrl mappings to avoid kernel panic.

I'm not sure that is the answer (unless you have tried it successfully).
The bcm2712 D0 stepping CPUs are all 64-bit, and in PINN this means they will use the overlays6 folder not overlays. bcm2712d0.dtb0 already exists there.
I do need to check the dtb files though as there are some differences there.

fprietog, 2025-07-22:
> Include bcm2712d0.dtbo in overlays folder so that D0 has the correct pinctrl mappings to avoid kernel panic. 

This doesn't work, at least for me.

I've changed cmdline.txt to loglevel=8 to get more info at boot:

[image]

Look at the list of the filesystems tried to mount root: ext3, ext2, ext4, vfat, msdos and f2fs... There is no squashfs in the list.

Maybe the kernel for PINN 3.9.7 was built without squashfs support, or maybe the ability to mount it as root was disabled...

This lead me to think that maybe the image of PINN 3.9.7 published is wrong. Has this concrete image published been tested in any Raspberry Pi 5 (or 4)?. After publishing I mean.

fprietog, 2025-07-22:
I've also googled for the _"No filesystem could mount, tried"_ message and I've found that a failed unzip of the root fs can lead to it. 

As I saw that you've updated pigz version, may this new version be faulty with D0 processor? You may built PINN 3.9.7 with old pigz version to try...

NumberOneGit, 2025-07-22:
If for now you try “device_tree=bcm2712d0-rpi-5-b.dtb” in config.txt do you still get the panic?

fprietog, 2025-07-22:
> If for now you try “device_tree=bcm2712d0-rpi-5-b.dtb” in config.txt do you still get the panic? 

Yes, same panic.

Here is a fix: I've replaced **kernel8.img** shipped in 3.9.7 with kernel8.img shipped in 3.9.5... and now 3.9.7 works (with the minute delay at OS boot as previous versions):

[image]

So:
3.9.5 kernel8.img md5sum: 07e97afd4e07f1b3da231634be42b8f3 OK
3.9.7 kernel8.img md5sum: 298fddb7e495b407536cce678a5f87d1 BAD

Maybe squasfs support has been dropped/broken in newest kernels...

procount, 2025-07-22:
Great!
Thanks for finding a solution.
Now I just have to find out why it went wrong...
It worked on my pi5.

procount, 2025-07-22:
@fprietog - could you raise the "minute delay before booting OS" as a separate issue, so it doesn't get forgotten?
Please include steps to reduce the issue & any configuration settings like forcetrigger being set.

fprietog, 2025-07-22:
> Great! Thanks for finding a solution. Now I just have to find out why it went wrong... It worked on my pi5.

Another difference between kernels is that newer PINN 3.9.7 kernel is 16k.

I'll test this PINN/kernel in a Raspberry Pi 4 that I'll borrow later.

> @fprietog - could you raise the "minute delay before booting OS" as a separate issue, so it doesn't get forgotten?
Please include steps to reduce the issue & any configuration settings like forcetrigger being set.

Ok, I'll do it. I'll try to include as much info I can because it just happened, there are no errors shown in dmesg or something like this.

procount, 2025-07-22:
Thanks for the report.
I'll probably just replace the kernel and reissue 3.9.7 for now and will try to sort out the build later as I will have little time over the next couple of weeks.

fprietog, 2025-07-22:
> Please try https://sourceforge.net/projects/pinn/files/archive/pinn-397a.zip/download

Just tested in "Rpi 5 16GB" and "Rpi 4 8GB", both working as expected*

_*just the one minute delay in the Rpi 5. I've opened an issue as we talked before but I think that it should be very difficult to fix unless you can reproduce it as no errors are shown._

smkrish007, 2025-08-09:
I followed above instruction for my RPi 5 16GB, it simply hangs every time it boots. Is there any other workaround which works on all RPi5 flavors ?

procount, 2025-08-09:
If you mean it "hangs" after PINN's splash screen is shown at about the time that the network is being established, then this is a known bug with PINN on the Pi5 -  See #770. This looks like an issue with Qt4.7 and the 64-bit kernel. It may need a complete redesign to use the later Qt version to work around this. For now, it is necessary to pull the power and reconnect until it works.

fprietog, 2026-02-11:
New PINN 3.9.8 version boots the Raspberry Pi 5 16GB (D0 stepping CPU) without problems!

Even the infamous boot minute delay (https://github.com/procount/pinn/issues/919) has gone!