procount/pinn #921: RPI 5 LibreELEC Version Update?

axertor, 2025-08-04:
Hello,

Is it possible to update the installer's LibreELEC version for RPI 5? The current 12.0.0 version is rather outdated from 5/4/24 and versions 12.0.2 and later appear to have a pretty critical mesa patch which LibreELEC is unusable without. I've tried the route of installing from USB but it's much more difficult than it needs to be..

a17suliman, 2025-10-02:
> +1 for this request, please...
> 
> [@axertor](https://github.com/axertor), in the meantime, i found a fairly easy workaround to update the os remotely here, i hope that helps! https://forum.libreelec.tv/thread/29766-rpi5-pinn-libreelec-12-batocera/

@scientificsound Thank you, have you successfully done this? I am trying but keep getting "Cannot write to" error since it looks like it's a read only file system.

procount, 2025-10-15:
Latest LibreELEC is now included.

closed 2025-10-15 by procount