procount/pinn #922: HDMI CEC not working - missing /dev/vchiq?

machacekmartin, 2025-08-30:
---
name: HDMI CEC not working - missing /dev/vchiq
about: Unable to use HDMI CEC during boot selection probably due to missing /dev/vchiq
---

**Describe the bug**
My goal was to use TV Remote during boot selection via HDMI CEC.
It doesn't work, so I went ahead and tried it inside the actuall installed OSs.
Both of them work with the remote, so TV is set up correctly, and it is supported. The only place it doesn't work is PINN GUI.

I then look into the debug file within CTRL+ALT+F2 terminal and see these three lines
```
CecListener thread started
Looking for "/mnt/cec_keys.json"
Error during vchi_initialise()
```

https://github.com/procount/pinn/blob/8335ec5c21456a26585b99fb328ba85c76469a9b/recovery/ceclistener.cpp#L114

= The ceclistener did not start at all.
It uses `vcgencmd` under the hood, but once I try `vcgencmd commands`, it gives `VCHI initialization failed
`. The `vcgencmd` tries to use `vchiq` binary, but upon closer clook at `/dev/` folder, it does not contain any `vchiq` at all. Am I doing something wrong? Shouldn't CEC work out of the box here?

**To reproduce**
Pretty much as described above

**Expected behaviour**
I'm able to control boot selection gui using my tv remote.

**Actual behaviour**
Boot selection does not respond to TV remote controls at all.

**System**
Add answers to the following questions:

* Which model of Raspberry Pi? 
Raspberry Pi 5 (4GB RAM)
* Which version of PINN?
3.9.3

**Logs**
```
CecListener thread started
Looking for "/mnt/cec_keys-json"
Error during uchi_initialise()
```

**Additional context**

procount, 2025-08-30:
CEC is not yet supported in PINN on the Pi5 due to the new kms drivers etc and 64 bit kernel. It needs completely rewriting.
It will work on all the lesser models though.

procount, 2025-08-30:
I don't have much time for PINN at all at the moment due to personal reasons - just enough to keep it going. The Pi5 is quite a different beast so there's a lot of work required to get it up to the same standard. It really needs re-architecting.
So I still hope to get CEC working along with the new DSI screens etc., but don't hold your breath. Maybe things will be easier next year.

machacekmartin, 2025-08-31:
Oh okay, do you know of any alternative "dual booter" I could use until the support for cec is added back for pi5 by any chance?

machacekmartin, 2025-08-31:
Nevermind! I just realised that I don't need to be able to decide which OS to boot into during PINN boot selection, all I need is 1 executable script for each of my OS which will reboot the system into the other OS.
No alternatives needed and no cec needed 😄 thank you for the hard word btw!

machacekmartin, 2025-09-02:
In the end I ditched all 3rd party multiboot solutions.
My entire goal was to have LibreELEC and Batocera in dual boot on PI5, while not having to reach for keyboard and mouse to switch between them (I don't want to use built-in Kodi in Batocera, because it doesn't have good support for 4k HDR).

For those wanting to do the same thing -- 
I managed to implement my own solution with Pi's autoboot.txt and custom partitions for OS separation.
Switching between OSs is done via custom .sh scripts that modify contents of autoboot.txt and call reboot. 
In LE they are invoked via menu button and in Batocera they are invoked from game "ports" section.

I'll keep this issue open for reference, since the problem with PI5 HDMI CEC in PINN still remains the same.